firstof9/ha-renogy #14: Renogy Logo's

Anto79-ops, 2024-12-15:
### The problem

Cosmetic suggestion to add the Renogy Logo to the integration :)

For example 

[image]

### What version of Home Assistant?

2024.12.3

### What version of Renogy?

1.0.0-b5

firstof9, 2024-12-16:
That has to go into the HA brand's repo, it's on the to-do list.

behold81, 2025-01-20:
https://github.com/home-assistant/brands/pull/6400

Done. hopefully one less item for you to worry about.

firstof9, 2025-01-20:
@behold81 thanks!