firstof9/ha-renogy #16: [Bug]: No battery Data Coming through

clennon81, 2024-12-16:
### The problem

The battery data is not coming through to the integration, as you can see in the screenshots, the One portal is showing the status, but integration only shows Unknown except for the status that shows as connected

[image]
[image]

### What version of Home Assistant?

core-2024.12.3

### What version of Renogy?

1.0.0-b7

### Diagnostics information

[config_entry-renogy-01JF76H31JMAAMQDSG9GR31K97.json](https://github.com/user-attachments/files/18147162/config_entry-renogy-01JF76H31JMAAMQDSG9GR31K97.json)
[home-assistant_renogy_2024-12-16T08-34-24.249Z.log](https://github.com/user-attachments/files/18147164/home-assistant_renogy_2024-12-16T08-34-24.249Z.log)

### Anything in the logs that might be useful for us?

```txt
To me it seems logs show "none" as the data for most of the entities.
```

### Additional information

it looks like its not pulling data from renogy, i guess that means its the API thats not working from renogy rather than an issue with the integration?

Anto79-ops, 2024-12-16:
according to your logs, the data is there, maybe just a tweak of the code to see them:

```
2024-12-16 08:32:51.943 DEBUG (MainThread) [custom_components.renogy] DEVICE: {'deviceId': '230803041918824003', 'name': 'Rover 40 Big Panels', 'mac': '10', 'firmware': '', 'status': 'online', 'connection': 'Bluetooth', 'serial': '', 'model': 'RNG-CTRL-RVR40', 'data': {'totalEnergyProduced': '327.36', 'batteryChargingAmps': '0.09', 'communicationMethod': None, 'batteryTemperature': '25', 'soc': '100', 'solarVolts': '14.4', 'loadAmps': '0.00', 'solarWatts': '0.86', 'batteryChargingVolts': '13.20', 'loadWatts': '0', 'sku': None, 'solarAmps': '0.06', 'batteryType': 'Lithium', 'loadVolts': '13.20'}}

```

firstof9, 2024-12-16:
The API reply is indicating `None` for your battery data, that's something Renogy will need to fix.
I can parse in however the extra solar data, the `solarAmps` are those measured in `A` or `mA` in the App?

Anto79-ops, 2024-12-16:
send an email to cc.chang@renogy.com about the none data. My inverter was similar, and a few hours later, they fixed it.

You can direct them to this issue, too.

firstof9, 2024-12-16:
Oops I forgot to also ask, is `totalEnergyProduced` in `kWh` or just `Wh` or are they using `Ah`?

clennon81, 2024-12-16:
in my one.renogy.com portal it shows as

Total Energy Produced
327.66 kWh

firstof9, 2024-12-16:
b8 should parse more sensors for you @clennon81

clennon81, 2024-12-16:
Thanks, I do see more data and sensors now showing a bit more info including total for one set of my panels

danc89, 2025-01-04:
Any further ideas to battery data not showing up? My 2 x 100ah lithium batteries are not pulling any data in. Also only one battery is showing in the integration?

clennon81, 2025-01-04:
> Any further ideas to battery data not showing up? My 2 x 100ah lithium batteries are not pulling any data in. Also only one battery is showing in the integration?

its a simple API problem, not the integration, Renogy clearly havnt added everything to the API yet properly. I believe they are working on an update soon according to discord

firstof9, 2025-01-05:
Keep poking the renogy dev on discord, they seem to get things added pretty quickly.

clennon81, 2025-01-10:
looks like renogy have fixed this now, all data showing
[image]

danc89, 2025-01-10:
Yup, data is now coming through. I have 2 batteries but the integration only gives me one battery device not the 2nd.

danc89, 2025-01-10:
The data is there in the diagnostics data but the integration only gives me one battery. Do I need to label the batteries different names to sort this out?

firstof9, 2025-01-10:
> The data is there in the diagnostics data but the integration only gives me one battery.

Can you post the data output here so I can review it?

danc89, 2025-01-10:
[renogy-01JH7S9XYFCT0DY1KBD8XNJWJA-RBT100LFP12S-G1-babd988cee5f2485fa7a54d492439e1d.json](https://github.com/user-attachments/files/18377692/renogy-01JH7S9XYFCT0DY1KBD8XNJWJA-RBT100LFP12S-G1-babd988cee5f2485fa7a54d492439e1d.json)
[image]

firstof9, 2025-01-10:
Only one device is being created for both batteries?

firstof9, 2025-01-10:
Ah I see the issue, Renogy's passing the same serial number for the batteries for some reason.

firstof9, 2025-01-10:
I'll see what I can figure out for ya.

Anto79-ops, 2025-01-10:
Your integration does pick up the second Mac address, though if you look at the drvice information, there are two listed.

danc89, 2025-01-10:
> Your integration does pick up the second Mac address, though if you look at the drvice information, there are two listed.

Yea I can see that, and the integration shows the 2 mac addresses on the left but it doesn't split them up into 2 different entities

firstof9, 2025-01-10:
You may need to delete the battery device(s) so they can be regenerated or remove/add the integration.

danc89, 2025-01-10:
> You may need to delete the battery device(s) so they can be regenerated or remove/add the integration.

Tried that, nothing changed.

firstof9, 2025-01-10:
Ok thanks for testing, I'll keep looking at it.

firstof9, 2025-01-10:
@danc89 I used your data to make some tests and got it working, please check out beta16 when you have a moment.

Thanks!

danc89, 2025-01-10:
> @danc89 I used your data to make some tests and got it working, please check out beta16 when you have a moment.
> 
> Thanks!

You're a wizard! That's worked perfectly, thank you.

closed 2025-01-10 by firstof9