firstof9/ha-renogy #19: [Bug]: Battery Present Ampers is in mA but data is in A

craigusus, 2024-12-16:
### The problem

I think you have the calculations for mA and A the wrong way around.
I know I have 2.4A coming in. If I change the unit to mA I get 2.4, but if I change it to A it shows 0.00248.

[image]
[image]

(Also, the spelling of Ampers should be Amperes or Amps)

### What version of Home Assistant?

core-2024.12.3

### What version of Renogy?

1.0.0-b9

craigusus, 2024-12-16:
No problem.
If you need anymore help/testing, I'm willing.

firstof9, 2024-12-16:
I would be great if they added the unit to their API output, perhaps emailing them would get them to add it.

Anto79-ops, 2024-12-16:
email: cc.chang@renogy.com

he's heading the API development and very will respond next day.

Anto79-ops, 2024-12-18:
@craigusus looks like a backend change was done at Renogy. Did this change your issue? try reloading the integration.

firstof9, 2024-12-18:
> looks like a backend change was done at Renogy

What do you mean?

Anto79-ops, 2024-12-18:
for consistent units

firstof9, 2024-12-18:
So they changed it to mA for all the ampere readings?

Anto79-ops, 2024-12-18:
thats what im trying to determine. Mine has not changed.

craigusus, 2024-12-18:
No mine are still in mA.
I did email cc.chang and ask if they could add something like the below:
```
    "data": {
        "communicationMethod": null,
        "presentCapacity": null,
        "error": null,
        "remainingTime": null,
        "heatingModeStatus": null,
        "temperature": "15",

        "temperatureUnit": "°C",

        "presentVolts": "12.80",
        "cellVolts": null,
        "maximumCapacity": "99.83",

        "maximumCapacityUnits": "%",

        "totalMaximumCapacity": "99.83",
        "firmwareVersion": "0026",
        "sku": null,
        "batteryLevel": "11.92",

        "batteryLevelUnits": "V",

        "presentAmps": "-0.15",

        "presentAmpsUnits": "A"
    }
```

craigusus, 2024-12-18:
But this is Renogy, they can't even quality control their own apps.

firstof9, 2024-12-18:
> No mine are still in mA.

So it's still reporting as `0.00251 A` ?

craigusus, 2024-12-18:
> > No mine are still in mA.
> 
> So it's still reporting as `0.00251 A` ?

That's correct.

scadman1, 2025-01-30:
Hi, I'm not sure if its related but I have a RBC50D1S-G4 and the solar charging amperes are a few of decimal places out. The DC App reads, 1.52A but I get 1,520.00 A in Home Assistant.

craigusus, 2025-01-30:
> Hi, I'm not sure if its related but I have a RBC50D1S-G4 and the solar charging amperes are a few of decimal places out. The DC App reads, 1.52A but I get 1,520.00 A in Home Assistant.

Correct. A is reading the mA value and mA is reading the A value.
Until the API states the correct values I have created a new sensor like the below to convert it:
```
renogy_solar_battery_1_present_ampers_a:
  friendly_name: "Renogy Solar Battery 1 Present Ampers (A)"
  unique_id: renogy_solar_battery_1_present_ampers_a
  unit_of_measurement: "A"
  device_class: current
  value_template: >-
    {% set renogy_battery = states('sensor.battery_1_present_ampers') | float(0) / 1000 %}
    {{ renogy_battery | round(2) }}
```

Anto79-ops, 2025-01-30:
If you haven't joined already, you may wish to join the Renogy API Discord Channel. This is where the Renogy Devs hang out, and with the little bit more voices, perhaps they'll fix this sooner.

https://discord.gg/VHVnDnY5

firstof9, 2025-03-27:
The unit should now be pulled from the API and parsed by the integration.

PR #60 (2025-03-27): feat: use unit of measurement from API data if available
fixes #19