firstof9/ha-renogy #29: [Bug]: Error contacting Renogy API: list index out of range

danc89, 2024-12-29:
### The problem

Unable to load after inputting keys

### What version of Home Assistant?

core-2024.12.5

### What version of Renogy?

1.0.0-b7

### Anything in the logs that might be useful for us?

```txt
Error contacting Renogy API: list index out of range
Traceback (most recent call last):
  File "/config/custom_components/renogy/config_flow.py", line 45, in async_step_user
    await renogy.get_devices()
  File "/usr/local/lib/python3.13/site-packages/renogyapi/__init__.py", line 119, in get_devices
    if "deviceId" in response[0].keys():
                     ~~~~~~~~^^^
IndexError: list index out of range
```

firstof9, 2024-12-30:
Use the latest beta. b9

danc89, 2024-12-30:
I have tried all the versions, all still the same

Anto79-ops, 2024-12-30:
I think @firstof9 fixed this [here](https://github.com/firstof9/py-renogy/pull/4) but no update was issued.

Anto79-ops, 2024-12-30:
https://github.com/firstof9/py-renogy/issues/3

firstof9, 2024-12-30:
Odd, it should have been in b9, try b11.

danc89, 2024-12-30:
Thanks for the help guys, its now saying "The API returned no devices for your account."
I'm not sure if i'm missing something. My DC home app will show me my batteries and and I think my M1 panel.

Anto79-ops, 2024-12-30:
yeah, make sure your M1 panel is paired with the devices. I know you can bypass the M1/ONE Core by adding devices directly to the app, but its only the M1/ONE Core hub that communucates with the API. When you go the the M1, do you see devices reporting to it and the email its connected to, is that the same email you used for the API keys account?

danc89, 2024-12-30:
Captain stupid here, m1 panel had un paired itself, all sorted now guys. Just waiting for it to actually connect to my batteries.
Thanks guys !

firstof9, 2024-12-30:
Someone previously had a similar issue so I wouldn't put yourself down by any means :slightly_smiling_face: 
Enjoy!

closed 2024-12-30 by firstof9