firstof9/ha-renogy #30: [Feature Request]: SoC values on Shunt300

jiresongit, 2025-01-02:
### The feature

At the moment the Shunt300 is not showing State of Charge values 
[image]

jiresongit, 2025-01-02:
"4834980418541485178": {
      "deviceId": "4834980418541485178",
      "name": "Shunt300",
      "mac": "xxxxxxxxxxxxxxxxxxxxxxxxxx",
      "firmware": "",
      "status": "online",
      "connection": "Mesh",
      "serial": "",
      "model": "RSHST-B02P300-G1",
      "data": {
        "current": -0.306,
        "starterBatteryVolts": 0.0,
        "temperature": -3276.800049,
        "sku": "RSHST-B02P300-G1",
        "batteryVolts": 13.410001
      }

firstof9, 2025-01-02:
No SoC is provided by the API.

Anto79-ops, 2025-01-02:
iteresting, looks like the API is not sending out that data, that would be the most critical for your shunt.

Send an email to cc.chang@renogy.com and/or join the Renogy API discord channel here 

https://discord.gg/GNp4pJ9t

Renogy still has this API as "beta" so any issues or feedback to them, they will action it.

firstof9, 2025-01-02:
I can however see about getting the `current` and `starterBatteryVolts` added.
can you show me what the app shows for this device?

jiresongit, 2025-01-02:
[image]

firstof9, 2025-01-02:
Oh boy their API's missing lots of stuff lol
Thanks!

Anto79-ops, 2025-01-03:
just curious, since Renogy aknowleged your request in the discord chat, did they add the SOC to the API already? They are pretty quick, sometimes. Try reloading the integration hopefully it shows up in the diagnostics

jiresongit, 2025-01-04:
This has now been added by renogy and is coming through to Ha A 😁👍

closed 2025-01-04 by jiresongit

Anto79-ops, 2025-01-04:
awesome!!