firstof9/ha-renogy #32: [Feature Request]: Implement switching of smart relays

jiresongit, 2025-01-02:
### The feature

To be able to switch on/ off smart relays via home assistant.

Currently in the smart scenes there is no way to adjust the relays based on current, voltage or location.

Being able to switch off items as power goes low, turning on relay controlling awning light (I use mine in a caravan) as you get near etc would be amazing!

firstof9, 2025-01-02:
There is no data available to toggle switches via the renogy API.
If this is a feature you'd like, it'd be better implemented by pairing zigbee devices to Home Assistant via Zigbee2MQTT or ZHA.

jiresongit, 2025-01-02:
> There is no data available to toggle switches via the renogy API. If this is a feature you'd like, it'd be better implemented by pairing zigbee devices to Home Assistant via Zigbee2MQTT or ZHA.

I figured as much. I want them paired to the renogy as my renogy is in a caravan but would be useful to use the power of HA at home if online.

firstof9, 2025-01-02:
I'll tag it as an API problem for now, until they release info how to toggle them.

behold81, 2025-01-20:
> I figured as much. I want them paired to the renogy as my renogy is in a caravan but would be useful to use the power of HA at home if online.

You could put a small RPI with HA in the van running the sensors in the van, then use the API Webhook call of HA to call it from your other HA. bonus you now have local HA in the van.......

scadman1, 2025-10-26:
Hi, did this get any traction as I would find this extremely useful also?

firstof9, 2025-10-26:
> did this feature gain any traction as I would find this extremely useful also?

No the API is read only for switches.