firstof9/ha-renogy #34: [Bug]: Sensors for Door Sensor

jiresongit, 2025-01-02:
### The problem

    "4836836027160995474": {
      "deviceId": "4836836027160995474",
      "name": "Door Sensor",
      "mac": "00124B0024533F3A",
      "firmware": "",
      "status": "offline",
      "connection": "Zigbee",
      "serial": "",
      "model": "DS01",
      "data": {
        "alarm": 1,
        "device name": "Door Sensor",
        "lowbattery": 100,
        "sku": "DS01"
      }

### What version of Home Assistant?

2024.12.5

### What version of Renogy?

1.0.0-b9

closed 2025-01-02 by firstof9

firstof9, 2025-01-02:
fixed in [b13](https://github.com/firstof9/ha-renogy/releases/tag/1.0.0-b13)

jiresongit, 2025-01-04:
Pls can you add the "alarm" also, this is the trigger of motion or door open.

PR #36 (2025-01-02): fix: add more sensors
fixes #31 
fixes #35
fixes #34