firstof9/ha-renogy #35: [Bug]: Sensors for Motion Sensor

jiresongit, 2025-01-02:
### The problem

    "4739063822287306028": {
      "deviceId": "4739063822287306028",
      "name": "Motion Sensor",
      "mac": "00124B00292DC212",
      "firmware": "",
      "status": "offline",
      "connection": "Zigbee",
      "serial": "",
      "model": "MS01",
      "data": {
        "alarm": 0,
        "device name": "Motion Sensor",
        "lowbattery": 73,
        "sku": "MS01"
      }

### What version of Home Assistant?

2024.12.5

### What version of Renogy?

1.0.0-b9

firstof9, 2025-01-02:
fixed in [b13](https://github.com/firstof9/ha-renogy/releases/tag/1.0.0-b13)

jiresongit, 2025-01-04:
Pls can you add the "alarm" also, this is the trigger of motion or door open.

firstof9, 2025-01-05:
It was already added in beta 13. Are you not seeing the binary sensor?

jiresongit, 2025-01-07:
[home-assistant_renogy_2025-01-07T10-52-04.965Z.log](https://github.com/user-attachments/files/18331420/home-assistant_renogy_2025-01-07T10-52-04.965Z.log)
Attached log, i've removed some "noise" from other components, but suspect perhaps the top two lines are the issue? It's trying to set the "alarm" as "status"? and "status" already exists? Looking through there are quire a few extra potential sensors across my other devices showing in the json that are not mapping through also.

firstof9, 2025-01-07:
Please try `b14` thanks.

closed 2025-01-07 by firstof9

jiresongit, 2025-01-07:
> Please try `b14` thanks.

Perfect thank you

PR #36 (2025-01-02): fix: add more sensors
fixes #31 
fixes #35
fixes #34
PR #38 (2025-01-07): fix: adjust door/window contact sensor name
fixes #35