firstof9/ha-renogy #40: [Bug]: Batteries showing `deviceId` as Serial number

craigusus, 2025-01-10:
### The problem

The integration is showing the `deviceId` as the Serial number:
[image]

`serial` has no data:
```
"240718093501078002": {
      "deviceId": "240718093501078002",
      "name": "RBT100LFP12S-G1",
      "mac": "30",
      "firmware": "",
      "status": "online",
      "connection": "Hub",
      "serial": "",
      "model": "RBT100LFP12S-G1",
      "data": {
        "communicationMethod": null,
        "presentCapacity": null,
        "error": null,
        "remainingTime": null,
        "heatingModeStatus": null,
        "temperature": "16",
        "presentVolts": "13.20",
        "cellVolts": null,
        "maximumCapacity": "99.97",
        "totalMaximumCapacity": "99.97",
        "firmwareVersion": "0026",
        "sku": null,
        "batteryLevel": "63.15",
        "presentAmps": "-0.21"
      }
```

### What version of Home Assistant?

core-2025.1.2

### What version of Renogy?

1.0.0-b15

### Diagnostics information

[renogy-01JF8HX9QWSA1NNVHYKNBP2Z7Q-Renogy ONE M1-9c81eff4a36e41120a761b7f926fffda.json](https://github.com/user-attachments/files/18380175/renogy-01JF8HX9QWSA1NNVHYKNBP2Z7Q-Renogy.ONE.M1-9c81eff4a36e41120a761b7f926fffda.json)

firstof9, 2025-01-10:
This is related to #16

firstof9, 2025-01-10:
Fixed in beta16

craigusus, 2025-01-10:
Fixed, thank you

closed 2025-01-10 by craigusus