firstof9/ha-renogy #42: [Question]: How is the 'Connected via' is worked out?

craigusus, 2025-01-10:
### The problem

Hi, not a major problem. I'm just curious about how this worked out.

My setup is as follows:

**Controller (RNG-CTRL-WND30)**
<- _RS232_ ->
**BT-1 Bluetooth Module**
<- _bluetooth_ ->
**ONE M1 (RMS-LP4-G2)**
<- _ethernet_ ->
**Battery 2 (RBT100LFP12S-G1)**
<- _ethernet_ ->
**Battery 1 (RBT100LFP12S-G1)**

But when I click on the Controller info it says it's connected via Battery 2:
[image]

The same with Battery 1 and the One M1:
[image]
[image]

They can't all be connected via the Battery. The Controller is connected to the One M1 via the Bluetooth module.

### What version of Home Assistant?

core-2025.1.2

### What version of Renogy?

1.0.0-b15

### Diagnostics information

[renogy-01JF8HX9QWSA1NNVHYKNBP2Z7Q-Renogy ONE M1-9c81eff4a36e41120a761b7f926fffda (1).json](https://github.com/user-attachments/files/18380533/renogy-01JF8HX9QWSA1NNVHYKNBP2Z7Q-Renogy.ONE.M1-9c81eff4a36e41120a761b7f926fffda.1.json)

firstof9, 2025-01-10:
This should also be sorted in beta16

craigusus, 2025-01-10:
> I notice your batteries are showing up as Hubs in your dump, that's very odd. Could you provide a full debug log so I can investigate?
> 
> Thanks.

Sure:
[home-assistant_renogy_2025-01-10T18-18-52.623Z.log](https://github.com/user-attachments/files/18380836/home-assistant_renogy_2025-01-10T18-18-52.623Z.log)

craigusus, 2025-01-10:
> I notice your batteries are showing up as Hubs in your dump, that's very odd. Could you provide a full debug log so I can investigate?
> 
> Thanks.

Sure:
[home-assistant_renogy_2025-01-10T18-18-52.623Z.log](https://github.com/user-attachments/files/18380857/home-assistant_renogy_2025-01-10T18-18-52.623Z.log)

firstof9, 2025-01-10:
`'connectType': ''` on the batteries, I'll have to add some extra logic into the library to make sure subdevices are never flagged as a hub.

Thanks!

firstof9, 2025-01-10:
Library has been updated, thanks!
I'll drop a new beta for this.

closed 2025-01-12 by craigusus