firstof9/ha-renogy #56: [Bug]: Controllers merged

craigusus, 2025-03-04:
### The problem

Hi again,

Ive just added a 2nd Contoller (RNG-CTRL-WND30) to my Renogy ONE M1 via a BT-1 module.
But the integration has merged the 2 Controllers into 1.

[image]

### What version of Home Assistant?

2025.2.5

### What version of Renogy?

1.0.3

### Diagnostics information

[renogy-01JNGY3J0XK4ASYQ3B5MADZF2B-Renogy ONE M1-9c81eff4a36e41120a761b7f926fffda.json](https://github.com/user-attachments/files/19073968/renogy-01JNGY3J0XK4ASYQ3B5MADZF2B-Renogy.ONE.M1-9c81eff4a36e41120a761b7f926fffda.json)

### Anything in the logs that might be useful for us?

```txt

```

firstof9, 2025-03-04:
Looks like your controllers are reporting the same `mac`:

```
      "name": "Controller Shed",
      "mac": "10",
```
```
      "name": "Controller House",
      "mac": "10",
```

That would be why they're getting merged.

craigusus, 2025-03-04:
Hmmm, not helpful.
It would be nice if it reported the full MAC address, not just the first 2 characters.

Thank you.

firstof9, 2025-03-04:
Should be solved in 1.0.4

closed 2025-03-04 by firstof9

craigusus, 2025-03-04:
Sorry, they're still merged but showing 2 MAC values.

[image]

firstof9, 2025-03-04:
You should be able to remove/delete the device and reload the integration.
Once they're merged like this in the device registry they stay that way until removed/rediscovered.

craigusus, 2025-03-04:
Ah yes, sorry. I thought I'd already done that! 😬 
Thank you again for jumping on it so quick!

craigusus, 2025-03-04:
Yes, both separate devices now.

[image]

[image]

firstof9, 2025-03-04:
Excellent, no problem. I just wanted to make sure it worked 100% as expected.
Thanks for confirming!

PR #57 (2025-03-04): fix: if duplicate network macs are found use the device_id
fixes #56