firstof9/ha-renogy #59: [Bug]: 'Failed setup, will retry: None' after HA OS update

craigusus, 2025-03-17:
### The problem

I just updated the HA OS and the integration is failing to load
[image]

I have tried with new keys, same result.

I tried to remove and re-add again, but now I have no integration.

### What version of Home Assistant?

core-2025.3.3

### What version of Renogy?

1.0.5

### Diagnostics information

[home-assistant_renogy_2025-03-17T22-28-39.163Z.log](https://github.com/user-attachments/files/19299945/home-assistant_renogy_2025-03-17T22-28-39.163Z.log)

### Anything in the logs that might be useful for us?

```txt

```

firstof9, 2025-03-18:
`renogyapi.exceptions.NotAuthorized` <-- there's your problem.

Your API keys are bad.

craigusus, 2025-03-18:
They were working fine up until the OS update/reboot.
And I have tried 3 different sets of keys since. Same result.

firstof9, 2025-03-18:
No idea, the error's coming from Renogy's servers as not authorized.

Anto79-ops, 2025-03-18:
just to confirm the basics:

1. Is the Renogy hub and HA instance connected to the internet?
2. Is the email account in the hub settings is associated to the same email account you used to create the API keys?
3. When you created the new keys, did you copy and paste the correct key in the correct text box for the integration? i.e. the `sk` key is the secret key and the `ak` key is the access key?

If all the above is yes, than yeah, somethings weird about this.

My integration is working after updating HAOS to 15 so could be a local issue, but im not sure myself

craigusus, 2025-03-18:
Yes
Yes
and Yes.

It was working perfectly fine up until the update/reboot. There is/was nothing wrong with the API keys or account.

I have just, again for the 8th time, deleted my API key from https://platform.renogy.com and recreated. Did a FULL system reboot of the Raspberry Pi, not just a HA restart, and tried again. This time it allowed me to add the integration back again!

The only other thing I did differently was name the key 'Renogy Home Assistant' as opposed to just 'Home Assistant'.

[image]

craigusus, 2025-03-27:
Sorry, I've just updated to v1.0.6 (then v1.0.7), and it's happened again.
There was, again, nothing wrong with the API keys before the update/restart of HA.
I did a full system reboot as per last time; I'm still getting errors and no integration.

firstof9, 2025-03-27:
Yup keep your keys, the renogy API auth is funky and I forgot it was per URL for some weird reason lol

firstof9, 2025-03-27:
Ok give 1.0.8 a go.

Anto79-ops, 2025-03-27:
Hey, it works! Thanks @firstof9 

Two of my entities have unit mismatch, though. 

Like KWh, A

craigusus, 2025-03-27:
Yes, 1.0.8 seems to be working again.
Thank you for jumping on so quick again.

Anto79-ops, 2025-03-27:
i think Renogy made some mistakes, @firstof9 can you confirm? 

```
  "totalKwhGenerated": [
          489691.0,
          "KWh"
        ],
```
and
```
        ],
        "outputAmps": [
          370.0,
          "A"
        ],
```

should be 0.37 A and 489.691 KWh

firstof9, 2025-03-27:
> can you confirm?

I can confirm it looks like they should be sending "mA" and "Wh" for your specific device.

Anto79-ops, 2025-03-27:
@firstof9, thanks. I just pinged them on the Discord chat. I'm sure they'll fix it.

Anto79-ops, 2025-03-27:
@craigusus sure, here you go!

https://discord.gg/HUMqueaj

closed 2025-03-31 by firstof9