firstof9/ha-renogy #63: [Bug]: 1.0.8 missing amp and soc

jiresongit, 2025-03-27:
### The problem

With API change and version uodatr shunt soc percentage and amps do not show.

Unavailable.

[image]

### What version of Home Assistant?

Recent

### What version of Renogy?

Newest

### Anything in the logs that might be useful for us?

```txt

```

firstof9, 2025-03-27:
Please attach the diagnostics file.

jiresongit, 2025-03-27:
[renogy-01JGKVP12VS8MGKSMNJPY4JWFP-Shunt300-d53b643704e68f0263130f8129cc77a7-1.json](https://github.com/user-attachments/files/19493736/renogy-01JGKVP12VS8MGKSMNJPY4JWFP-Shunt300-d53b643704e68f0263130f8129cc77a7-1.json)

firstof9, 2025-03-27:
```json
      "name": "Shunt300",
      "mac": "2500B301001E009F284974E14C000100",
      "firmware": "",
      "status": "online",
      "connection": "Mesh",
      "serial": "",
      "model": "RSHST-B02P300-G1",
      "data": {
        "starterBatteryVolts": [
          0.0,
          "V"
        ],
        "temperature": [
          -3276.800049,
          "\u00b0C"
        ],
        "power": [
          -0.643,
          "A"
        ],
        "batteryVolts": [
          13.219001,
          "V"
        ]
      }
```
The API isn't reporting a SOC or an amperage reading.

firstof9, 2025-03-27:
Please enable debug logging so we can take a look at what's coming down the pipe please.

jiresongit, 2025-03-27:
So looks like the ampere one is now called power, but doesn't come thru... And no soc? [error_log-3.log](https://github.com/user-attachments/files/19494033/error_log-3.log)

firstof9, 2025-03-27:
> So looks like the ampere one is now called power, but doesn't come thru... And no soc?

That seems to be it, "power" doesn't make sense for an ampers reading either. I would suggest poking the Renogy devs on discord, especially since your SoC has gone missing, maybe they borked something on the backend.

behold81, 2025-04-01:
This is happening on mine too. Both ajunta failing to report SoC. 

I would assume many will get this issue.

firstof9, 2025-04-02:
> This is happening on mine too.

Make sure to report it on their discord if you haven't already.

Anto79-ops, 2025-04-02:
@behold81 here is the link to Renogy's Discord channel. You can report the error directly to their devs:

https://discord.gg/HmgN7hbz