firstof9/ha-renogy #64: [Bug]: New units arent valid

craigusus, 2025-03-27:
### The problem

The new units (°C and KWh) arent valid units for Energy and Temperature

### What version of Home Assistant?

2025.3.4

### What version of Renogy?

1.0.8

### Diagnostics information

[home-assistant_renogy_2025-03-27T21-14-42.162Z.log](https://github.com/user-attachments/files/19494399/home-assistant_renogy_2025-03-27T21-14-42.162Z.log)

### Anything in the logs that might be useful for us?

```txt

```

craigusus, 2025-03-27:
Just mentioned it on the Discord server too. It seems they have done a lot of messing up of correct unit spellings.

Anto79-ops, 2025-03-27:
You're right. There are plenty of mistakes, but I know they'll fix it within a couple of days they're pretty responsive on Discord

craigusus, 2025-03-27:
I guess that's the difference between Developers and Customer Service Agents (I've had a few bad experiences with the latter).

firstof9, 2025-03-27:
It's fine, I still have the "home assistant way" of units for fallback.
1.0.9-b0 will be up for testing this adjustment shortly.

firstof9, 2025-03-27:
I only fixed the ones in this log here, where there others?

craigusus, 2025-03-27:
I found these further up the log too:
```
2025-03-27 19:25:10.545 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.controller_house_battery_temperature (℃) cannot be converted to the unit of previously compiled statistics (°C). Generation of long term statistics will be suppressed unless the unit changes back to °C or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2025-03-27 19:25:10.545 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.controller_house_total_energy_produced (KWh) cannot be converted to the unit of previously compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2025-03-27 19:25:10.545 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.controller_shed_battery_temperature (℃) cannot be converted to the unit of previously compiled statistics (°C). Generation of long term statistics will be suppressed unless the unit changes back to °C or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2025-03-27 19:25:10.546 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.controller_shed_total_energy_produced (KWh) cannot be converted to the unit of previously compiled statistics (kWh). Generation of long term statistics will be suppressed unless the unit changes back to kWh or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2025-03-27 19:25:10.746 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.renogy_battery_1_maximum_capacity (AH) cannot be converted to the unit of previously compiled statistics (Ah). Generation of long term statistics will be suppressed unless the unit changes back to Ah or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2025-03-27 19:25:10.746 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.renogy_battery_2_maximum_capacity (AH) cannot be converted to the unit of previously compiled statistics (Ah). Generation of long term statistics will be suppressed unless the unit changes back to Ah or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
```

firstof9, 2025-03-27:
"AH", added that now too.

firstof9, 2025-03-27:
Beta released, let me know if you run into any other weirdness.

craigusus, 2025-03-28:
Sorry, I switched back to 1.0.8. The beta was still showing incorrect values, especially 'mA' vs. 'A'.
```
"presentAmps": [
          0.14,
          "A"
```
1.0.8: (correct but still showing 'AH' not 'Ah')
[image]

1.0.9-b0: (showing mA values but 'A' unit. still showing 'AH' not 'Ah') 
[image]

firstof9, 2025-03-28:
> incorrect values, especially 'mA' vs. 'A'.

I think this one needs to be reported to Renogy.

craigusus, 2025-03-28:
Abel messaged me directly. So I made a suggestion that ALL the units are correct and standardised across ALL devices.

[image]

Anto79-ops, 2025-03-28:
@craigusus thank you! I was just talking about this attention to detail.

Anto79-ops, 2025-03-31:
Renogy did good, and fixed the units issues (for me at least) a few hours ago:

`'loadVolts': (0.0, 'V')` previously this was unit W
`'outputAmps': (370.0, 'mA')` - previosly this was unit A
`'totalKwhGenerated': (492008.0, 'Wh')` - previosly this was KWh

I'm running 1.0.9-b2 now. The HA database may complaing about unit change but your can easily fix that by going to statitics and fix issue.

craigusus, 2025-03-31:
So close...

Anto79-ops, 2025-03-31:
yes, you're right, just the temperature unit now. Probably soon!

craigusus, 2025-05-11:
I think we're all good on this for now?

PR #65 (2025-03-27): fix: adjust unit of measurements to comply with home assistant
fixes #64
closed 2025-05-11 by craigusus