firstof9/ha-renogy #66: [Bug]: Only importing devices from the first Renogy ONE

MADNYXX, 2025-03-27:
### The problem

I have 2 Renogy ONEs monitoring 2 different systems. The addon has worked perfectly to import the first Renogy ONE devices but ignores the second.

I would be happy to assist in further testing / troubleshooting on this issue. I could set up an additional API key temporarily for devs to play with if that helps.

[image]

[image]

### What version of Home Assistant?

Core-2025.3.4

### What version of Renogy?

1.0.5

### Diagnostics information

HA logs only show attempting to add devices from NAVI.

[image]
NOTE: Error is due to those devices already successfully being added

### Anything in the logs that might be useful for us?

```txt

```

firstof9, 2025-03-27:
I have no idea about secondary hubs @Anto79-ops does a second hub need a separate key pair?

Anto79-ops, 2025-03-27:
oh thats a great question. To be honest, I dont know.  How does it show up in the app? two hubs?

This a great question and may have to update the docs when figured out.

firstof9, 2025-03-27:
@MADNYXX can you post the diagnostics and maybe gather some debug logs too?

MADNYXX, 2025-03-27:
I have just updated to 1.0.8 with same issue. I have also tested with multiple API keys. 

Looking at the logs during the GET DEVICE/LIST API call it returns both Renogy ONES and all their associated devices as sublists:

2025-03-28 09:49:05.864 DEBUG (MainThread) [renogyapi] Response: [{'deviceId': '92058263905894401', 'sn': '**************', 'sku': 'RSHGWSN-W02W-G1', 'name': 'NAVI', 'category': 'Renogy ONE Core', 'mac': '4C:E1:74:46:CD:6B', 'firmware': 'V1.1.194', 'onlineStatus': 'online', 'connectType': '', 'addTime': 1721009823000, 'sublist': [{'deviceId': '4696380053888015553', 'sn': '', 'sku': 'RBC50D1S-G1', 'name': 'DC-DC', 'category': 'DC-DC', 'mac': 'F8:55:48:17:53:9E96', 'firmware': '0.0.103', 'onlineStatus': 'online', 'connectType': 'bt', 'addTime': 1729583727000}, {'deviceId': '4742927428989540063', 'sn': '', 'sku': 'RSHST-B02P300-G1', 'name': 'NAVI', 'category': 'Battery Shunt', 'mac': '2500B301001E0004944C74E14C000100', 'firmware': '', 'onlineStatus': 'online', 'connectType': 'mesh', 'addTime': 1738195965000}, {'deviceId': '4766223760552351282', 'sn': '', 'sku': 'RBT100LFP12S-G1', 'name': 'Battery', 'category': 'Battery', 'mac': 'C4:D3:6A:62:CE:A248', 'firmware': 'V1.16', 'onlineStatus': 'online', 'connectType': 'bt', 'addTime': 1738298214000}, {'deviceId': '4864113760936419486', 'sn': '', 'sku': 'RNG-CTRL-RVR40', 'name': 'Solar', 'category': 'Controller', 'mac': 'C4:D3:6A:5D:D6:9316', 'firmware': '2.0', 'onlineStatus': 'online', 'connectType': 'bt', 'addTime': 1724055094000}]}, {'deviceId': '151846673048406117', 'sn': ''**************',', 'sku': 'RSHGWSN-W02W-G2', 'name': 'Gerry', 'category': 'Renogy ONE Core', 'mac': '4C:E1:74:57:A3:C0', 'firmware': 'V1.1.194', 'onlineStatus': 'online', 'connectType': '', 'addTime': 1741143199000, 'sublist': [{'deviceId': '4658184205058089965', 'sn': '', 'sku': 'RBT12200LFP-BT', 'name': 'RBT12200LFP-BT', 'category': 'Battery', 'mac': '4C:E1:74:59:AB:1D48', 'firmware': 'V1.16', 'onlineStatus': 'online', 'connectType': 'bt', 'addTime': 1741143303000}, {'deviceId': '4681970704843382129', 'sn': '', 'sku': 'RBT12200LFP-BT', 'name': 'RBT12200LFP-BT', 'category': 'Battery', 'mac': '4C:E1:74:59:A9:BB48', 'firmware': 'V1.16', 'onlineStatus': 'online', 'connectType': 'bt', 'addTime': 1741143371000}, {'deviceId': '4745763454457569407', 'sn': '', 'sku': 'RNG-CTRL-ROVER60', 'name': 'Solar 60', 'category': 'Controller', 'mac': 'C4:D3:6A:5D:DF:9816', 'firmware': '2.0', 'onlineStatus': 'online', 'connectType': 'bt', 'addTime': 1742611445000}, {'deviceId': '4761701246413007011', 'sn': '', 'sku': 'RBT12200LFP-BT', 'name': 'RBT12200LFP-BT', 'category': 'Battery', 'mac': '4C:E1:74:59:A6:7C48', 'firmware': 'V1.16', 'onlineStatus': 'online', 'connectType': 'bt', 'addTime': 1741143371000}, {'deviceId': '4800224501887819119', 'sn': '', 'sku': 'RNG-CTRL-RVR40', 'name': 'Solar 40', 'category': 'Controller', 'mac': 'C4:D3:6A:5D:5B:C016', 'firmware': '2.0', 'onlineStatus': 'online', 'connectType': 'bt', 'addTime': 1741143371000}, {'deviceId': '4846677662840600646', 'sn': '', 'sku': 'RBC50D1S-G4', 'name': 'DC-DC', 'category': 'DC-DC', 'mac': 'C4:D3:6A:5D:FD:4096', 'firmware': '0.0.103', 'onlineStatus': 'online', 'connectType': 'bt', 'addTime': 1741143681000}]}]

It looks like the component is taking the first device ID (92058263905894401 name:NAVI in my case) and listing all device data associated with that device. No further reference to the second Renogy ONE (id: 151846673048406117 name:Gerry) in the rest of the logs.

Anto79-ops, 2025-03-27:
I do see the two hubs? Is the Rover 60 and Rover 40 connected to Gerry?

I don't see any data, though, coming for either Hub, but maybe that's just in the Diagnostics

MADNYXX, 2025-03-27:
[config_entry-renogy-01JQCTDWG9ZAKCF7TPWG74MP00.json](https://github.com/user-attachments/files/19495727/config_entry-renogy-01JQCTDWG9ZAKCF7TPWG74MP00.json)

[error_log-2.log](https://github.com/user-attachments/files/19495812/error_log-2.log)

Yes, both Renogy ONEs show in the logs during initial device/list. Yes, Rover 40 & 60 belong to Gerry.  All expected devices are there during this logs action.

I have attached the DEBUG logs. I noticed this part:
2025-03-28 09:50:56.506 DEBUG (MainThread) [custom_components.renogy] Finished fetching (Renogy CoreR9da5ccbeb8344d0380cb79204589b769kcxt2yyll1qe1fz8fc2o7wttnjwg4hp2qkqsdcolo6zlt0f5kidvk4cky6mmdiaplyslvwfpgl1sbcos63icoq8g7q6o6dw19sfv) data in 8.297 seconds (success: True)
2025-03-28 09:50:56.506 DEBUG (MainThread) [custom_components.renogy] Looking for hub device...
2025-03-28 09:50:56.506 DEBUG (MainThread) [custom_components.renogy] Hub found: 92058263905894401
2025-03-28 09:50:56.506 DEBUG (MainThread) [custom_components.renogy] DEVICE: {'deviceId': '92058263905894401', 'name': 'NAVI', 'mac': '4C:E1:74:46:CD:6B', 'firmware': 'V1.1.194', 'status': 'online', 'connection': 'Hub', 'serial': '', 'model': 'RSHGWSN-W02W-G1', 'data': {}}
2025-03-28 09:50:56.506 DEBUG (MainThread) [custom_components.renogy] Using device: 92058263905894401 via None

This doesn't happen for the second hub. I assume once a Hub is found it continues rather than appending for each.

firstof9, 2025-03-28:
I'll see what I can figure out. Thanks for the logs.

firstof9, 2025-03-28:
I know what the issue is, I wasn't accounting for multiple hubs. I'll see what I can do to get it sorted.

MADNYXX, 2025-03-28:
Awesome, thanks! I really appreciate the effort on this custom integration!

If you need me to give you an API key to test with, let me know.

firstof9, 2025-03-28:
Please test `1.0.9-b1` this should handle your multi-hub situation. Thanks!

MADNYXX, 2025-03-28:
Updated version is working perfectly! 
I didn't even have to reconfigure the integration. New devices just automatically populated.

Thank you so much! 👍

PR #67 (2025-03-28): fix: handle multiple renogy hub devices
fixes #66
closed 2025-03-28 by firstof9