firstof9/ha-renogy #69: [Feature Request]: Alternator Charging Information

scadman1, 2025-04-25:
### The feature

In addition the the existing solar watts and amps it would be great to be able to monitor the charge coming from the alternator on my RBC50D1S-G4.

scadman1, 2025-04-26:
Yes, it's Renogy's latest version of their 50amp DC-DC with MPPT. It shows me the MPPT info just not the DC-DC side.

firstof9, 2025-04-26:
Can you download the diagnostics and upload the file to a comment here?

jiresongit, 2025-04-26:
I have the dcdc30 which is the baby brother of the dcdc50 the starter battery volts and amps which contain the values you want aren't exposed by the api currently

Anto79-ops, 2025-04-26:
If it's an API limitation, you may wish to join the Renogy Dev Discord chat and ask for them to expose the values.

https://discord.gg/VjTFV3vY

I would still post your diagnostics. Perhaps your model is different, but you never know.

scadman1, 2025-05-27:
Hi @Anto79-ops , I have had confirmation from Abel on Discord that this information is now available from the API. I would like to assist with testing but unfortunately don't have the skill set to help with the code implementation.

Anto79-ops, 2025-05-27:
Oh that's great news it's awesome when Renogy makes a change based on customer feedback.

Try reloading the integration and see if the entities show up. If they don't download the diagnostic and post it here. First will probably be able to add those on next release

Anto79-ops, 2025-05-27:
go to settings, integrations, click on the ">" for renogy and click on the 3 dots, and click on diagnostics:

[image]

should download a file after a few seconds.

scadman1, 2025-05-27:
Is that what you need?

firstof9, 2025-05-27:
yes, but for your Alternator device

scadman1, 2025-05-27:
[image]

firstof9, 2025-05-27:
Looks like they're reporting to me.
From your diagnostics file:
```json
        "auxiliaryBatteryChargingWatts": [
          19.0,
          "W"
        ],
        "auxiliaryBatteryChargingVolts": [
          13.8,
          "V"
        ],
```
and:
```json
        "startBatteryAmps": [
          60.0,
          "A"
        ],
        "startBatteryVolts": [
          12.7,
          "V"
        ],
```
Your screen shot from Home Assistant shows values as well. You should be good to go.

Anto79-ops, 2025-05-27:
Keep an eye on how it updates. Maybe Renogy crossed the streams in the backend.

firstof9, 2025-05-27:
The unit of measure comes from the API, if the unit's incorrect, you'll have to poke Able on discord.

scadman1, 2025-05-28:
I've reported the returned amp issue to the Renogy developers and asked them to look into it. Apologies for any confusion caused earlier, the initial returned figures were all over the place. I can confirm it has now started returning data only when the engine is running its just the data received is incorrect, as per @firstof9 states this must be a Renogy issue.