firstof9/ha-renogy #70: [Feature Request]: Full support for Rego 60A DC-DC charger

alphadogfpv, 2025-04-26:
### The feature

If possible, please add full support for RBC1260DO-12B (60A 'Rego' battery to battery charger).

I have this connected (via BT) to a Renogy One M1 (in addition to a Rego 60A MPPT charger also connected to the M1 which is working great!).

Currently the DC-DC charger is kind of connecting, but is definitely not making all the data on the One M1 (or DC Home) available in HA.

On DC Home, which has multiple tabs, I see:
Status - (at top but not labelled, app shows a lozenge below the graphic) with say "CURRENT LIMIT" or "NOT CHARGING" or something shown
Input tab:
- Starter Battery Volts  (eg 14.6 V)
- Starter Battery Amps (e.g. 49.02 A)
Output tab:
- Aux Battery Charge Watts (e.g. 668 W)
- Aux Battery Volts (e.g. 13.9 V)
- Aux Battery Charge Amps (e.g. 48.07 A)
- Aux Battery Temperature (e.g. 25 ℃ - although I don't have a probe connected so that is just some default)
-Others tab:
- DC-DC Temperature (eg 27 ℃)
- Total kWh generated (e.g. 4.65 kWh)
- Battery Type (e.g. "User" - corresponds to the mode dial on the front of the unit)

On M1 home screen I see - unlabelled but what looks like:
- Aux Battery Charge Watts (e.g. 668.45 W)
- Aux Battery Charge Amps (e.g. 48.09 A)
Then when I click on DC-DC to show details I see:
A lozenge below the graphic - although when I checked mine was only showing " . . . "
- Battery Type (e.g. "User")
- Aux Battery Volts (e.g. 13.9 V)
- Starter Battery volts  (eg 14.6 V)
I haven't explored the M1 much, so maybe it shows more, but the above are the obvious ones.

btw - I updated HA-Renogy today to (1.09) and saw some changes:
- Units corrected for Aux Battery Charge Watts (previously showed 0.00 A; now shows 0.00 W)
- Total Energy Generated (I think) previously said 4.65 kWh (which ties up with DC-Home; but now says 0.00465 kWh!  So this was working (I think although I can't be 100% certain of the units) but now x1000 to small.

I downloaded the diagnostics file, I'm not sure what you need and what is specific to me, but the DC-DC message part is below - please let me know if you ned more:

"##################": {
      "parent": "##################",
      "deviceId": "##################",
      "name": "DC-DC",
      "mac": "60",
      "firmware": "V2.0",
      "status": "online",
      "connection": "Bluetooth",
      "serial": "BT-TH-#######",
      "model": "RBC1260DO-12B",
      "data": {
        "auxiliaryBatteryTemperature": [
          25,
          "\u00b0C"
        ],
        "dc-dcMpptTemperature": [
          28,
          ""
        ],
        "communicationMethod": [
          null,
          ""
        ],
        "starterBatteryVolts": [
          12.5,
          "V"
        ],
        "totalKwhGenerated": [
          4.65,
          "kWh"
        ],
        "auxiliaryBatteryChargingWatts": [
          0.0,
          "A"
        ],
        "solarInputVolts": [
          0.0,
          "V"
        ],
        "auxiliaryBatteryVolts": [
          13.3,
          "V"
        ],
        "batteryType": [
          "User",
          ""
        ],
        "solarInputAmps": [
          0.0,
          "A"
        ],
        "auxiliaryBatteryChargingAmps": [
          0.0,
          "A"
        ],
        "starterBatteryAmps": [
          0.28,
          "A"
        ]
 
 Many thanks

alphadogfpv, 2025-04-26:
Sorry, I forgot to mention, currently on HA I see:
- Aux Battery charging Watts (but this is always 0.00 even when One M1 shows 700W or so!)
- Aux Battery Temperature (e.g. 25 ℃ - correct)
- Total Energy Generated (e.g. 0.00465 kWh - looks like data is there, just scaling issue)
- Battery Type (e.g. "User") - correct
- Connection Type ("Bluetooth")
- Status ("Connected")

firstof9, 2025-04-26:
> * Units corrected for Aux Battery Charge Watts (previously showed 0.00 A; now shows 0.00 W)
> * Total Energy Generated (I think) previously said 4.65 kWh (which ties up with DC-Home; but now says [...]

This is how the API reports it, you'll need to have Renogy fix that.

> "dc-dcMpptTemperature": [
>           28,
>           ""
>         ],

This measurement is missing the unit, also needs to be reported to Renogy.

alphadogfpv, 2025-04-26:
Thank you for the quick response.

tbh I'm not that fussed about the temperature units, or even the scaling of total energy generated kWh, but I will raise it with Renogy if I can find some mechanism to do so - please let me know if you have a link or know the right approach

What would be brilliant though would be support for 
1. Aux Battery Charge Watts, which is currently always showing as 0.00, and in the diagnostic message above it says:
   "auxiliaryBatteryChargingWatts": [0.0, "A"]  (so is auxiliaryBatteryChargingAmps)- does this mean this is a Renogy API issue too?  The charger was running when I downloaded diagnostics, but I don't know if the example data message is from that moment or from  earlier when the charger was not charging.

2. Starter Battery volts would also be really useful - it does look like this is populated in the message

thanks again

firstof9, 2025-04-26:
> 1. "auxiliaryBatteryChargingWatts": [0.0, "A"]  (so is auxiliaryBatteryChargingAmps)- does this mean this is a Renogy API issue too?

Yes this is the what the API is sending out.

Here is the Renogy discord: https://discord.gg/VjTFV3vY

> 2\. tarter Battery volts would also be really useful

This I can add, looks like some devices report "startBatteryVoltage" and "startBatteryAmps" rather than `starter`. I'll get a PR setup for this in a moment.

WS-Dave, 2025-08-21:
alphadogfpv: I worked with Renogy (via their Discord channel() to get auxiliaryBatteryChargingWatts switched to Watts (at least for my RBC50D1S-G1... hopefully it fixed it for your 60A DC-DC

They also got the units set for "dc-dcMpptTemperature"
It is coming in from the API to the Integration - but is not reporting.

Chris: Is there something I need to expose/publish it?  Or is this in the code

(see below pasted from my "Downlod Diagnostics")
As you can see - it is using the same Unit Type as "auxiliaryBatteryTemperature" (which is reporting fine)

Thank you!!

   "230712210728093001": {
      "parent": "230309223001942120",
      "deviceId": "230712210728093001",
      "name": "DC-DC",
      "mac": "60",
      "firmware": "V99",
      "status": "online",
      "connection": "Unknown",
      "serial": "RBC50D1S-G1",
      "model": "RBC50D1S-G1",
      "data": {
        "auxiliaryBatteryTemperature": [
          25,
          "\u00b0C"
        ],
        "dc-dcMpptTemperature": [
          23,
          "\u00b0C"
        ],

firstof9, 2025-08-21:
I don't think i have a sensor defined for `dc-dcMpptTemperature`, please add a new issue feature request for this. Thanks.