firstof9/ha-renogy #71: [Question]: Device disconnected/dead but integration still getting data

craigusus, 2025-05-04:
### The problem

Hi again,
My ONE M1 decided to die, again. (This is the replacement one I received less than a year ago!).
It has been disconnected for about 30 minutes, but the integration is still getting data somehow?

### What version of Home Assistant?

core-2025.4.4

### What version of Renogy?

1.0.9

### Diagnostics information

[renogy-01JPN03XSM9MYGHEFVQDSVMZ91-Renogy ONE M1-9c81eff4a36e41120a761b7f926fffda.json](https://github.com/user-attachments/files/20027017/renogy-01JPN03XSM9MYGHEFVQDSVMZ91-Renogy.ONE.M1-9c81eff4a36e41120a761b7f926fffda.json)

### Anything in the logs that might be useful for us?

```txt

```

Anto79-ops, 2025-05-04:
Unfortunatly, my ONE Core also blew up a couple of weeks ago. It would not turn on and just make a squealing sound. It was warrantly replaced my Renogy so I thank them for that. I received a warranty replacement just a couple of days ago, and back online. I had a G1 version and received a G3 version. On the G3, I noticed they changed power requirements, i.e. it used to be DC input voltage of 8 to 54V, now its 8 to 36V. This probably means they did some major re-work on the power supple of the device. So, I hope it fairs better than my first gen device.

While my device was dead and offline, the numbers of the integration were still there, but were not updating. It basically just read the last number reported before its dead.

Are your numbers updating while the device is in this bad state?

firstof9, 2025-05-04:
Yes I didn't setup the sensors to go unavailable if the device is offline yet.

craigusus, 2025-05-04:
@Anto79-ops  Yes, the last time my M1 died was due to Renogy pushing out a duff firmware, and after jumping through many hoops, they finally sent a warranty replacement. I'm guessing they've done it again. I'm hoping this time the process will be quicker!
And the values aren't changing, so picking up the last values.

@firstof9 No problem; I just wanted to check which end the issue was at. I'd probably not have noticed much until this evening when the sun moved off my panels to see that I'm still receiving power in the dark. Except I was out and went to show someone the (poor) DC Home app and noticed the M1 was offline. Got home to find it just showing the Renogy boot logo, again!

craigusus, 2025-05-11:
Got my new M1 within a few days, so I can't really complain about that side of things. It was a bit of a pain to remove and re-add it back into HA, though. I had to remove > reboot > add (that said it had failed) > reboot > remove > reboot > add > reboot before it started working again.

Abel did message me to say, "_The firmware of M1 is updated infrequently. We recommend that you consider using the Renogy ONE CORE. It's the successor to M1 and a product we'll support in the long run_." I responded saying that I do not plan on purchasing any more Renogy products, and for me to get a ONE Core, Renogy will have to send me one as a warranty replacement.

closed 2025-05-11 by craigusus