firstof9/ha-renogy #72: [Bug]: Error when trying to add integration

JerseyJames, 2025-05-11:
### The problem

This integration sounds great but I'm getting an error when trying to add it.
I have obtained API keys, downloaded the Custom Integration using HACs and rebooted Home Assistant.
I then go to add the Integration. 'Renogy' is found in the list. I leave the Device Name set to the default of 'Renogy Core' and enter the Secret Key and Access Key. On clicking 'Submit' I get 'An unknown error has occurred, please check your Home Assistant log file for details.'
On doing so I see this:

`This error originated from a custom integration.

Logger: custom_components.renogy.config_flow
Source: custom_components/renogy/config_flow.py:50
integration: Renogy (documentation, issues)
First occurred: 1:29:07 PM (2 occurrences)
Last logged: 1:30:09 PM

Error contacting Renogy API: 'count'
Traceback (most recent call last):
  File "/config/custom_components/renogy/config_flow.py", line 50, in async_step_user
    await renogy.get_devices()
  File "/usr/local/lib/python3.13/site-packages/renogyapi/__init__.py", line 164, in get_devices
    data["connection"] = SUBDEVICE_CONNECTION_TYPE[
                         ~~~~~~~~~~~~~~~~~~~~~~~~~^
        device["connectType"]
        ^^^^^^^^^^^^^^^^^^^^^
    ]
    ^
KeyError: 'count'`

Can you suggest where I'm going wrong? The only thing I can think of is that maybe I need to pay for the Renogy subscription? I was on the one month trial a while back and I could see the data from my Renogy One Core.. but the trial is no longer still active.

Many thanks in advance, James

### What version of Home Assistant?

core-2025.5.1

### What version of Renogy?

1.0.9

### Anything in the logs that might be useful for us?

```txt
This error originated from a custom integration.

Logger: custom_components.renogy.config_flow
Source: custom_components/renogy/config_flow.py:50
integration: Renogy (documentation, issues)
First occurred: 1:29:07 PM (2 occurrences)
Last logged: 1:30:09 PM

Error contacting Renogy API: 'count'
Traceback (most recent call last):
  File "/config/custom_components/renogy/config_flow.py", line 50, in async_step_user
    await renogy.get_devices()
  File "/usr/local/lib/python3.13/site-packages/renogyapi/__init__.py", line 164, in get_devices
    data["connection"] = SUBDEVICE_CONNECTION_TYPE[
                         ~~~~~~~~~~~~~~~~~~~~~~~~~^
        device["connectType"]
        ^^^^^^^^^^^^^^^^^^^^^
    ]
    ^
KeyError: 'count'
```

firstof9, 2025-05-11:
Show the error from your `home-assistant.log` file, not from the UI.

JerseyJames, 2025-05-11:
Certainly, Chris. Here it is. Many thanks....

`2025-05-11 19:50:18.869 ERROR (MainThread) [custom_components.renogy.config_flow] Error contacting Renogy API: 'count'
Traceback (most recent call last):
  File "/config/custom_components/renogy/config_flow.py", line 50, in async_step_user
    await renogy.get_devices()
  File "/usr/local/lib/python3.13/site-packages/renogyapi/__init__.py", line 164, in get_devices
    data["connection"] = SUBDEVICE_CONNECTION_TYPE[
                         ~~~~~~~~~~~~~~~~~~~~~~~~~^
        device["connectType"]
        ^^^^^^^^^^^^^^^^^^^^^
    ]
    ^
KeyError: 'count'
`

JerseyJames, 2025-05-12:
Currently a Renogy One Core, a 50A DC/DC Charger with MPPT (RBC2125DS-21), 2 x 200Ah Smart Lithium batteries (soon to be 3) and a 3Kw Inverter (though I don't believe this device reports to the One Core.

If you think it might be device-related, for some reason, ever since install, the One Core has also detected a Victron Cerbo GX which isn't mine and I'm pretty sure can't possibly be in range (which is odd).

Regards,

James

How many devices do you have?

firstof9, 2025-05-12:
I'll drop a beta shortly for you to test that should solve this.

firstof9, 2025-05-12:
[Beta](https://github.com/firstof9/ha-renogy/releases/tag/1.0.10-b0) is release, please see if that resolves this error message you're getting.

JerseyJames, 2025-05-12:
Thank you Chris. I have downloaded and installed the beta and it works perfectly. It is showing all 5 devices (including the Victron Cerbo GX and excluding the Inverter (as expected)). Well done.
If it's not too tedious - what did you have to do to fix it?
Also, if you've got a second could you tell me how often your code polls for the data and (if you know) how often the data is uploaded to the Renogy cloud from the Renogy One Core? (as it all seemed a bit slow when I was using their trial)
Kind regards, James

PS. Do reach out to me directly if I can help with any further testing with any of the devices that I have that you may not have access to.

Anto79-ops, 2025-05-12:
oh so interesting, you can see a Victron product on the Renogy API/hub?
Could you please show a screenshot of this.

JerseyJames, 2025-05-12:
> oh so interesting, you can see a Victron product on the Renogy API/hub? Could you please show a screenshot of this.
> 
> I wonder what other victron products show up?

I'm actually away from home at the moment, Anto79-ops, so I can't show a picture of the hub and I can't show a screenshot of the Portal as I'm not currently subscribed (because it annoys me that Renogy charge for this when others (like Victron) don't... and because it's too expensive for my taste. I'll try to remember to include a picture when I'm home later in the week.
As I say I have no idea where the Cerbo GX device came from. I think it was coincidental that I was playing with a friends Cerbo GX only a day or so before I plugged the Renogy One Core in as they were not in the same place as each other and not on any connected networks etc. I live quite remotely with only one neighbour that could be within Bluetooth range and I doubt he's got any Victron kit... so it's a mystery. I haven't deleted it as I'm keeping an eye on it to see if it ever reports any data - it hasn't so far (in about 5 months).
Actually, I have just found a couple of old pics on my phone showing the Cerbo - one of the One Core and one of the phone app. (Attached)

[image]

[image]

Anto79-ops, 2025-05-12:
lol I can't believe that. I wonder if this was an accident.

thank for sharing this!

firstof9, 2025-05-12:
> If it's not too tedious - what did you have to do to fix it?

In the library I just added some extra handling for if the `connectType` isn't something I've seen before or listed in the Renogy API. If you download the diagnostics in the integration and attach them to a comment here, I can show you where it's wonky.

closed 2025-07-25 by firstof9