firstof9/ha-renogy #87: [Bug]: Missing data fields on 200ah batteries.

Sparhawk76, 2025-09-09:
### The problem

I have 6 200 AH batteries monitored by my Renogy One M1, 3 older ones with Bluetooth, and 3 of the newer PRO series.  In both cases I don't get the Present Capacity metric published to HA. 

The metrics that do show up are: 
Battery Level 
Maximum Capacity
Present Amperes
Present Voltage
Temperature

### What version of Home Assistant?

core-2025.9.1

### What version of Renogy?

Version 1.0.13

### Diagnostics information

[renogy-01K4NP3DQH7DWBAKT1CQ1724RJ-Renogy ONE M1-80004b4c1d293c64426294caabc498be (1).json](https://github.com/user-attachments/files/22237696/renogy-01K4NP3DQH7DWBAKT1CQ1724RJ-Renogy.ONE.M1-80004b4c1d293c64426294caabc498be.1.json)

### Anything in the logs that might be useful for us?

```txt

```

firstof9, 2025-09-09:
> Present Capacity metric

Do you have an example of this from the Renogy app?

I don't see anything called "present capacity" in the API data.

Sparhawk76, 2025-09-09:
No, wasn't me.  Haven't been on the Renogy Discord.

Anto79-ops, 2025-09-09:
here you go:

https://discord.gg/QpSdRztf

closed 2025-11-04 by firstof9